thp/psmoveapi #449: Upload Java bindings to Maven Central

DeflatedPickle, 2022-08-11:
Hey! Thanks for your work on the library! If possible, it'd be nice to have the Java wrapper on Maven Central so that it may be downloaded and updated from build tools like Maven or Gradle. One of the easiest ways to do this is via [JitPack](https://docs.jitpack.io/building/#gradle-projects)

rafaellago, 2022-09-14:
I like this Idea, and been thinking how to make this possible with little fuss to the end user as possible. The main problem being the native libraries. Because it would have to be packaged a different one for every build possible (windows, mac, Linux) and now that different processors architectures are more widespread things get even more difficult (arm, x86/64 and all). So one could make a distributable jar containing only the "most used" native(x86/64 win/linux and m1 macs) and give a hard time for raspberrys or other arm SBCs around there. 

Even if all possible builds were to be made available, there would be cases left out, which goes against the WORA philosophy of Java, and you would end up with a bloated jar with lots of unnecessary distributions of the same lib.

We could make a base jar with only java binaries and have the dev download and distribute only the needed native libs. But I think this would cause more harm than good.

For the moment, I still think that loading the lib from the outside is still better than having it as a maven dependency. 

But it's good to know that there are other Java Devs interested in the library and hopefully someone may come with a better idea how to overcome those doubts.

thp, 2022-09-17:
Closing as won't fix, as @rafaellago is right in the sense that we need to ship native libraries, and while we can probably support 95% of the users by providing x86_64 libraries for macOS, Windows, Linux, it's usually not that easy, especially with different macOS versions and different Linux distros.

closed 2022-09-17 by thp