thp/psmoveapi #450: M1 Pro Ventura Beta 4 - C compiler errors/Id:warnings with QTKit.tbd

CliveRosfield, 2022-09-01:
Psmove pair fails to pair but it gives me the controller address, so i disabled the bluetooth, ran "pgrep blued" till nothing came back. Manually Authorized the controller's MAC address using command "sudo defaults write /Library/Preferences/com.apple.Bluetooth HIDDevices -array-add "aa-bb-cc-dd-ee-ff" (my controller address wrote in)... I assume it accepts it because it doesn't give any complaint... Well after reconnecting the bluetooth and pressing the home button nothing happens, the controller doesn't show in my list of available devices to connect to at all. Also no prompt about a pin or anything, It acts as if it's not even trying to find it.... So I wondered if anything didn't fully do right along the way, even if says it built to 100%, so I'm wondering if the warnings about QTKit.tdb during building or the cmake error output complaining about "CMakeCCompilerId.c" and"CMakeCXXCompilerId.cpp" failing is causing the problem. Here's pictures of the end of the build and the picture after is of the cmake error output.
[image]
[image]
I'd really appreciate help with this, I've been trying to get this to work on my m1 for months now, took a break and came back after the recent fix from a few days ago for M1 Macs. I hope it can work on Ventura as I really like Ventura's controller profiles and all.

thp, 2022-09-02:
Have you tried with a fresh checkout from Git (latest master branch, which includes some M1-related fixes) and the script `scripts/macos/build-macos`? I verified it works on a M1 Max on Monterey, haven't tested with Ventura.

From the screenshots it looks like you're also using XCode beta? Linker errors like `-lSystem` not found and `-lc++` not found sound like something is wrong with your XCode/toolchain installation. Also, can you upload the full cmake/compiler logs as text file? The screenshots don't tell the whole story.

CliveRosfield, 2022-09-03:
So here is the CMakeError.log and CMakeOutput.log  ... I also included pictures of all the warnings that we're highlighted in yellow since I didn't do so before..
[CMakeOutput.txt](https://github.com/thp/psmoveapi/files/9481871/CMakeOutput.txt)
[CMakeError.txt](https://github.com/thp/psmoveapi/files/9481872/CMakeError.txt)
[image]
<img width="1274" alt="2nd set of warnings" src="https://user-images.githubusercontent.com/1
[image]
12556739/188257879-0f30dfee-4579-44bc-a0fa-06495ad9bd5f.png">

thp, 2022-09-08:
Random Google search brings up this:

https://www.intel.com/content/www/us/en/developer/articles/troubleshooting/error-ld-library-not-found-with-macos-big-sur.html

Quoting from there:

> a)    Install the Xcode command-line utilities:   xcode-select --install
> b)    Run: sudo xcode-select -s /Applications/Xcode.app/Contents/Developer
> c)    You may be prompted to accept a license.  If so, accept the license with: sudo xcodebuild -license

Alternatively, can you install Monterey on a USB and see if you can reproduce the issue? Again, Monterey + M1 Max works fine for me (full XCode installed).

thp, 2022-10-20:
Ventura now has a release date, let's see if it's still an issue then or not.

CliveRosfield, 2022-10-31:
It's still doing the same for me on Ventura, If it ends up working for you then I'd love to know.  I'm highly interest in playing all of RE Village with my move controllers!

thp, 2022-11-25:
Have you installed XCode completely, including its command-line tools, and have you launched it at some point so that it can install any additional components it might need?

thp, 2022-12-05:
On Ventura, I had to run this command to get command-line build tools working (make sure you update the Xcode app if you have it installed and it hasn't been updated yet):

```
xcode-select --install
```

thp, 2022-12-05:
Also don't forget that you might need to re-run `cmake` with a clean build directory and possibly re-build or re-install any dependencies with the new toolchain.

thp, 2022-12-17:
I was able to successfully build it using `scripts/macos/build-macos` on the final release version of macOS Ventura 13.1 using Xcode 14.2 as the toolchain/SDK. Closing as fixed.

closed 2022-12-17 by thp