thp/psmoveapi #452: Issues when reading magnetometer calibration

nitsch, 2022-10-13:
i get this message in my CECH-ZCM1E unit

```[PSMOVE CRITICAL] Assertion fail in psmove_load_magnetometer_calibration: result == 3```

is ok?

greetings

_Originally posted by @sl1pkn07 in https://github.com/thp/psmoveapi/issues/372#issuecomment-1275064852_

nitsch, 2022-10-13:
It means that there were problems reading the calibration data that was written to a file previously. Can you share the calibration file for closer inspection?

The file is named `xx_xx_xx_xx_xx_xx.magnetometer.csv` (with your controller's BDADDR substituted for the `xx`) and should be located inside the `.psmoveapi` directory below your home directory.

sl1pkn07, 2022-10-13:
ok, i thik i found the problem

if launch the demos/labs/etc with my locale (LANG=es_ES.UTF-8) fails with the message posted above. but if launch with LC_ALL=C or LANG=C, the message is gone

the file: [00_06_f7_d2_d8_36.magnetometer.csv](https://github.com/thp/psmoveapi/files/9781050/00_06_f7_d2_d8_36.magnetometer.csv)

greetings

thp, 2022-10-14:
Alright, this is because `fscanf()` used in `psmove_load_magnetometer_calibration()` and `fprintf()` used in `psmove_save_magnetometer_calibration()` are locale-dependent (`LC_NUMERIC`).

However, the command-line tools (e.g. the calibration tool) don't set the locale (hence the `.` in the file), but the GUI toolkits usually set it (hence the `,` expected by `*scanf` functions).

Some things that you could test:

1. Only setting `LC_NUMERIC="C"` + launching the GUI tools
2. Adding the following call in `psmove_init()`

```diff
diff --git a/src/psmove.c b/src/psmove.c
index f5db775..44c0eed 100644
--- a/src/psmove.c
+++ b/src/psmove.c
@@ -327,6 +327,8 @@ psmove_get_half_frame(PSMove *move, enum PSMove_Sensor sensor,
 enum PSMove_Bool
 psmove_init(enum PSMove_Version version)
 {
+    setlocale(LC_NUMERIC, "C");
+
     // Compile-time version of the library is PSMOVE_CURRENT_VERSION
     // Compile-time version of the user code is the value of "version"
 
```

As the locale affects all of the program, it's not ideal, as it means any program pulling in PS Move API would suddenly see numeric value with "." unconditionally, independent of locale. One option would be to use `newlocale()` + `uselocale()` just around the `*scanf()` and `*printf()` calls and restore the locale again after those calls. This is basically what macOS (+*BSD's) `*scanf_l()` functions provide (unfortuntely, those don't seem to be available in glibc).

See also: https://stackoverflow.com/a/13919957/1047040

thp, 2022-10-14:
Proposed PR in #453, @sl1pkn07 please give it a try if it fixes the issue for you.

sl1pkn07, 2022-10-19:
yes. #453 seems fixed my problem. the message is gone. and seems fixed a calibration problem, after the PR, i get this message when launch the paint  lab (for example)

~~~
[PSMOVE WARNING] Only 7/9 fields read from magnetometer calibration, discarding values. Please recalibrate.
~~~

after the recalibration, all OK

greetings

commit 5de54fe0 (2022-10-27): Locale-independent magnetometer calibration (Fixes #452)
commit 8a1f8d03 (2022-10-27): Merge pull request #453 from thp/thp/locale-independent

Make magnetometer calibration independent of locale (Fixes #452)
closed 2022-10-23 by nitsch