thp/psmoveapi #454: how to solve this calibration problem?

lotsarfati, 2022-10-18:
[image]

this is in the same calibration place, anywhere in the room i get the same result. there is any way to calibrate them in the same space? it looks like they move the same way unequally in my room.

nitsch, 2022-10-18:
I am not exactly sure what we are looking at here, but it seems to me
that this is just an issue of using the wrong origin in the associated
controller models. Align the model to have its origin at the center of
the orb and your renders match the photos quite nicely.

If this is not what you are getting at, please provide more details.

lotsarfati, 2022-10-19:
> I am not exactly sure what we are looking at here, but it seems to me that this is just an issue of using the wrong origin in the associated controller models. Align the model to have its origin at [...]

lotsarfati, 2022-10-19:
> I am not exactly sure what we are looking at here, but it seems to me that this is just an issue of using the wrong origin in the associated controller models. Align the model to have its origin at [...]

from top to bottom you see what happens in real life compers to what the computer see. so when they are opposite to one another they have huge distance but when the lights touching each other it works fine .
this problem is making all kind of wierd bugs because they move not at the same way in space .

i have another problem which i think that it is because of this one . when i sit and use kinnect to vr it align perfectly but when i stand up it suddenly stop being align and they are in unrealistic distance from one another

nitsch, 2022-10-23:
This is not what I meant. So let's try again:

What system are you using this in? How is the tracker and sensor data from psmoveapi fed into whatever you are using? These pictures alone are not of much use.

I assume you are using some optical tracker to determine the controller position. These trackers typically track the glowing orb, but your 3D models seem to assume that the position returned by the tracker is the middle of the controller, not the orb. Like I said, this looks to me like a misaligned 3D model, not a calibration issue in the library.

lotsarfati, 2022-10-24:
> This is not what I meant. So let's try again:
>
> What system are you using this in? How is the tracker and sensor data from psmoveapi fed into whatever you are using? These pictures alone are not of [...]

 what you mean by system? it is ps cameras and psmove with  psmoveapi program.

how i fix misaligned 3d model?

nitsch, 2022-10-26:
The screenshots do not look like any program we are shipping with the
library. At least nothing that I recognize. So what exactly is it that
you are using?

> how i fix misaligned 3d model?

By changing the point of its origin, I assume.

thp, 2022-12-17:
@lotsarfati It seems like the tool you are using isn't part of PS Move API, so not sure how much we can do about it here in this project.

As an aside, don't forget that the sensors are not in the controller's center, but slightly off (@nitsch had some teardown pictures at some point that showed where the sensors are), in case that's relevant. Add to this that the glowing sphere is on one side of the controller, and you have to do some additional offsetting between the global translation (from the controller tracked via the camera) and the global (local) rotation (from the controller's sensor inputs).

Closing this for now, as it's more of a support request for an external tool than PS Move API itself. Please reopen if there's something actionable for PS Move API here.

closed 2022-12-17 by thp