thp/psmoveapi #455: OpenCV 4.x and Qt5/Qt6 port

sl1pkn07, 2022-10-23:
Hi

any plans for update OpenCV to v4.x and Qt4 (EOL) to Qt5 or Qt6?

also, if i remember correctly, Bluez 5 van pair PS3 controllers without sixad (https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/plugins)

greetings

thp, 2022-10-27:
No immediate plans to update to newer OpenCV or Qt, but patches (tested on all 3 target OSes) are welcome.

thp, 2022-10-27:
> also, if i remember correctly, Bluez 5 can pair PS3 controllers without sixad (https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/plugins)

Sixaxis pairing is slightly different from Move pairing the last time I checked. If you mean Navigation Controller support, feel free to send an update to `docs/navcon.rst`. The documentation and tool should still work as-is, even for older BlueZ versions.

sl1pkn07, 2022-11-26:
for my part. with linux (qt5 port):

accelerometer3d

~~~
diff --git a/examples/labs/accelerometer3d/accelerometer3d.pro b/examples/labs/accelerometer3d/accelerometer3d.pro
index d1909be..2ae6f25 100644
--- a/examples/labs/accelerometer3d/accelerometer3d.pro
+++ b/examples/labs/accelerometer3d/accelerometer3d.pro
@@ -3,6 +3,8 @@ TEMPLATE = app
 DEPENDPATH += .
 INCLUDEPATH += .
 
+isEqual(QT_MAJOR_VERSION, 6): error("Requires Qt4 or Qt5. QGLWidget needs ported to QOpenGLWidget for work with Qt6")
+isEqual(QT_MAJOR_VERSION, 5): QT += widgets
 QT += opengl
 
 SOURCES += main.cpp
@@ -12,8 +14,17 @@ HEADERS += view.h
 
 HEADERS += orientation.h
 
-DEPENDPATH += ../../../include
-INCLUDEPATH += ../../../include
-
-LIBS += -L../../../build/ -lpsmoveapi -lGLU -lglut
-
+mac {
+    QT_CONFIG -= no-pkg-config
+    PKG_CONFIG = /usr/local/bin/pkg-config
+}
+CONFIG += link_pkgconfig
+packagesExist(psmoveapi) {
+    PKGCONFIG += psmoveapi
+} else {
+    DEPENDPATH += ../../../include
+    INCLUDEPATH += ../../../include
+    LIBS += -L../../../build/ -lpsmoveapi
+}
+PKGCONFIG += glu
+PKGCONFIG += glut
diff --git a/examples/labs/accelerometer3d/main.cpp b/examples/labs/accelerometer3d/main.cpp
index 1fa32d2..689b13e 100644
--- a/examples/labs/accelerometer3d/main.cpp
+++ b/examples/labs/accelerometer3d/main.cpp
@@ -27,8 +27,13 @@
  * POSSIBILITY OF SUCH DAMAGE.
  **/
 
+#include <QtGlobal>
 
+#if QT_VERSION < QT_VERSION_CHECK(5, 0, 0)
 #include <QtGui>
+#else
+#include <QApplication>
+#endif
 
 #include "orientation.h"
 #include "view.h"
diff --git a/examples/labs/accelerometer3d/view.cpp b/examples/labs/accelerometer3d/view.cpp
index 669f5d4..330fd25 100644
--- a/examples/labs/accelerometer3d/view.cpp
+++ b/examples/labs/accelerometer3d/view.cpp
@@ -167,7 +167,7 @@ renderController(QQuaternion quaternion, int buttons, int trigger)
     QMatrix4x4 mat;
     mat.rotate(quaternion);
     glScalef(4, 4, 4);
-    glMultMatrixd(mat.data());
+    glMultMatrixd((GLdouble*)mat.data());
 
 
 
~~~

needs to be ported from QGLWidget to QOpenGLWidget for work with Qt6 (sorry, im not coder)
funny: i have less warnings with Qt5 (zero) than Qt4 xd

thp, 2022-12-01:
Neat, want to turn that into a Pull Request?

commit 5d04d5d5 (2022-12-21): Quick and dirty Qt 5 porting (closes #455)
closed 2022-12-23 by thp