thp/psmoveapi #456: Unable to pair PSMove controller to M1 Macbook Air (Ventura)

BJS1206, 2022-12-03:
Hi, I am trying to pair a PSMove controller to my M1 Macbook Air (running MacOS Ventura) in order to play JS Joust on Sportsfriends but I've been struggling to get it to work. I followed the user-uploaded guide on steam [here](https://steamcommunity.com/app/277850/discussions/0/3150808675335075369/) and everything worked arrording to the guide up until the ```sudo defaults write``` part. When compiling psmoveapi myself and running ```sudo ./psmove pair```, this is what I get:

```
bryce@Bryces-Air build % sudo ./psmove pair
Password:
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in 04:76:6e:0f:0a:18 not yet calibrated.
PSMove #1 connected via USB.
[PAIRING OSX] No need to add entry for OS X before 10.7.
Pairing of #1 failed.
Calibration data available and saved.
bryce@Bryces-Air build % 
```

And so I run the MacOS troubleshooting pairing steps on [the documentation](https://psmoveapi.readthedocs.io/en/latest/pairing.html#mac-os-x) but when I enable bluetooth again the controller doesn't connect. Also, the bluetooth settings page doesn't show the move controller as a remembered device. I'm not really sure what else I should be doing. If you have anything I can try or do, please let me know, it would be greatly appreciated.

thp, 2022-12-05:
This seems suspicious:
```
[PAIRING OSX] No need to add entry for OS X before 10.7.
```
Let me check the behavior on my machine.

thp, 2022-12-05:
My machine outputs the following for [`sw_vers`](https://ss64.com/osx/sw_vers.html) with `-productVersion`:

```
13.0.1
```

Related: #32 (commit 7fa815021dfbdd70fc8095adf97ce4dcd8f5bd54).

It seems like this might have been broken since macOS 11 (Big Sur, 2020).

Fix has now been pushed to the `master` branch.

thp, 2022-12-05:
(I don't have a PS Move controller here to test this for real, so please reopen the issue if even after building with the latest master branch revision + trying to pair again it doesn't work.)

commit c492bacd (2022-12-05): Fix macOS version detection for macOS 11 and newer (fixes #456)
closed 2022-12-05 by thp