thp/psmoveapi #457: (Continuation of #456 ) Unable to pair on M1 Mac, Ventura

BJS1206, 2022-12-06:
Continuation of issue #456 (I don't think I can reopen an issue unless I closed it myself? Sorry, I'm not too familiar with GitHub.) 

Thank you for looking into it! It looks like that fixed the problem of detecting the wrong MacOS version (The ```[PAIRING OSX] No need to add entry for OS X before 10.7.``` message no longer appears), but unfortunately I still can't get my controller to pair. I suspect that it may have to do with the mac address of the controller being written to the ```com.apple.bluetooth``` file in ```/Library/Preferences/``` , is it possible that file it needs to be written to changed in recent MacOS versions? I'm thinking that because none of my other bluetooth devices are in that file, in fact, there's not much data there at all. Just a guess though, you would have a much better idea of what's causing the problem than I would.

One thing that I think is worth noting is the controller works well with the example scripts when it's plugged in over usb, the issue is just pairing over bluetooth.

[Here](https://vimeo.com/778278234) is a video of me trying to pair the controller to my Mac, in case you notice anything strange there.

Log from terminal:

```
bryce@Bryces-Air build % sudo ./psmove pair
Password:
Connected controllers: 1
[PSMOVE WARNING] Magnetometer in 04:76:6e:0f:0a:18 not yet calibrated.
PSMove #1 connected via USB.
[PAIRING OSX] Detected: macOS 13.1
[PAIRING OSX] Entry for 04-76-6e-0f-0a-18 already present.
Pairing of #1 succeeded!
Controller address: 04:76:6e:0f:0a:18
Calibration data available and saved.
bryce@Bryces-Air build % 
```

thp, 2022-12-12:
Let's mark this as broken in Ventura for now.

Bryceshaw06, 2023-02-07:
Hey Thomas, I just came across something that might be worth looking into related to this problem. I dug around a bit and noticed that all of my bluetooth devices had their information and mac addresses stored in different files like ```/Users/bryce/Library/com.apple.bluetooth.services.cloud/CachedRecords/DeviceRecord/01:23:45:67:89:10```

I am still not even sure if this is what should be looking for in order to help get this issue solved. I would like to help however I can, I just don't know what I should be doing. If you can tell me how to can help you best, I'd be more than happy to do whatever!

Krish120003, 2024-11-09:
I'd love to have this solved as well, currently on Sonoma and can't pair my controller.

clemon79, 2025-05-13:
Same issue in Sequoia. Really want to get this working again. I note JSJ is finding the controllers and making the appropriate entry files in .psmoveapi in my home folder via its own setup pridess, it's just the BT setup on the Mac side that is not working.

thp, 2025-06-13:
If someone wants to have a play with the Pi Pico W, there's example code here:

https://github.com/raspberrypi/pico-examples/blob/master/pico_w/bt/hid_host_demo/CMakeLists.txt
https://github.com/bluekitchen/btstack/blob/master/example/hid_host_demo.c

Alternatively, running PS Move API on Linux on any Raspberry Pi device and using `moved` to export that device to the macOS host would also work (or use a USB Bluetooth dongle, forward the USB device to a VM running Linux, and run PS Move API inside that VM).

thp, 2025-06-15:
Possibly this already exists: https://github.com/juan518munoz/PicoSwitch-WirelessGamepadAdapter