thp/psmoveapi #458: Successfully paired controller but LED keeps flashing

pgrenon1, 2022-12-14:
Hi, I'm on linux, Ubuntu 22.04, on a Surface Laptop (not the tablet, the one with a keyboard)

I have managed to pair a controller with `sudo ./psmove pair`. 
- I see it as Motion Controller in my bluetooth settings
- It is properly listed when I do `sudo ./psmove list`
- It starts vibrating and flashing and all when I do `sudo ./psmove responsiveness`.

However, the small red LED light keeps flashing. Is that normal? If I understand correctly, it should stop flashing and stay on?

Thanks in advance

thp, 2022-12-14:
If it otherwise connects and works, a blinking LED suggests its battery is
low. Charge the controller fully and the LED should stay on while connected.

pgrenon1, 2022-12-14:
Aaah I see, thanks!

closed 2022-12-14 by pgrenon1