thp/psmoveapi #485: Cannot compile on m1 Mac

Marshungry, 2023-02-10:
Tried compiling the source code on an M1 Mac but no luck. 

```
(base) arhammahajan@Arhams-MacBook-Pro psmoveapi % bash -e -x scripts/macos/build-macos
+ '[' 0 -eq 0 ']'
+ echo 'Usage: scripts/macos/build-macos [x86_64|arm64]'
Usage: scripts/macos/build-macos [x86_64|arm64]
+ exit 1
```

This is all that happens. Been on it for over a day and can't figure out the problem.

thp, 2023-02-10:
It says it right there:

```
Usage: scripts/macos/build-macos [x86_64|arm64]
```

You have to pass in either `x86_64` or `arm64` as parameter, depending on whether you have an Intel Mac or an Apple Silicon Mac.

For Intel Macs, use:

```
bash -e -x scripts/macos/build-macos x86_64
```

For Apple Silicon Macs, use:

```
bash -e -x scripts/macos/build-macos arm64
```

Marshungry, 2023-02-10:
Thank you so much. Finally got it working!!

closed 2023-02-10 by Marshungry