thp/psmoveapi #487: Build on M1 Ventura (13.3.1) fails

jango-fx, 2023-05-06:
Trying to build on a M1 MacBook Pro to no avail. 😭 
No idea what else could be done....

```
ld: warning: directory not found for option '-L/opt/homebrew/lib'
ld: library not found for -lusb
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [psmove] Error 1
make[1]: *** [CMakeFiles/psmove.dir/all] Error 2
make: *** [all] Error 2
```
[complete terminal output](https://github.com/thp/psmoveapi/files/11413284/build.log)

thp, 2023-05-06:
Please read the documentation, especially the part about installing dependencies:

https://psmoveapi.readthedocs.io/en/latest/build.html#building-on-macos-13

jango-fx, 2023-05-07:
i did. i tried. no idea what else i can do.

jango-fx, 2023-05-07:
that's the output of
```
bash -e -x scripts/install_dependencies.sh
```
```
+ set -e
+ set -x
++ uname
+ UNAME=Darwin
+ case "$UNAME" in
+ brew update
==> Updated Homebrew from c1fdb5908 to 112081237.
Updated 3 taps (homebrew/core, homebrew/cask and homebrew/cask-fonts).
==> Outdated Formulae
dav1d               exiftool            ffmpeg              ghostscript         imagemagick         jpeg-xl             leptonica           libheif             libomp              node                qt@5                rav1e               rubberband          svt-av1
==> Outdated Casks
arduino-ide                                              blender                                                  cryptomator                                              dotnet-sdk                                               powershell

You have 14 outdated formulae and 5 outdated casks installed.
You can upgrade them with brew upgrade
or list them with brew outdated.
+ brew install --force cmake git libtool automake autoconf python libusb-compat sdl2
Warning: Treating cmake as a formula. For the cask, use homebrew/cask/cmake
Warning: cmake 3.26.3 is already installed and up-to-date.
To reinstall 3.26.3, run:
  brew reinstall cmake
Warning: git 2.40.1 is already installed and up-to-date.
To reinstall 2.40.1, run:
  brew reinstall git
Warning: libtool 2.4.7 is already installed and up-to-date.
To reinstall 2.4.7, run:
  brew reinstall libtool
Warning: automake 1.16.5 is already installed and up-to-date.
To reinstall 1.16.5, run:
  brew reinstall automake
Warning: autoconf 2.71 is already installed and up-to-date.
To reinstall 2.71, run:
  brew reinstall autoconf
Warning: python@3.11 3.11.3 is already installed and up-to-date.
To reinstall 3.11.3, run:
  brew reinstall python@3.11
Warning: libusb-compat 0.1.8 is already installed and up-to-date.
To reinstall 0.1.8, run:
  brew reinstall libusb-compat
Warning: sdl2 2.26.5 is already installed and up-to-date.
To reinstall 2.26.5, run:
  brew reinstall sdl2
+ brew unlink libtool
Unlinking /Users/felix.groll/homebrew/Cellar/libtool/2.4.7... 20 symlinks removed.
+ brew link --overwrite libtool
Linking /Users/felix.groll/homebrew/Cellar/libtool/2.4.7... 20 symlinks created.
+ pip3 install --user sphinx
Requirement already satisfied: sphinx in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (7.0.0)
Requirement already satisfied: sphinxcontrib-applehelp in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (1.0.4)
Requirement already satisfied: sphinxcontrib-devhelp in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (1.0.2)
Requirement already satisfied: sphinxcontrib-jsmath in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (1.0.1)
Requirement already satisfied: sphinxcontrib-htmlhelp>=2.0.0 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (2.0.1)
Requirement already satisfied: sphinxcontrib-serializinghtml>=1.1.5 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (1.1.5)
Requirement already satisfied: sphinxcontrib-qthelp in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (1.0.3)
Requirement already satisfied: Jinja2>=3.0 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (3.1.2)
Requirement already satisfied: Pygments>=2.13 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (2.15.1)
Requirement already satisfied: docutils<0.20,>=0.18.1 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (0.19)
Requirement already satisfied: snowballstemmer>=2.0 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (2.2.0)
Requirement already satisfied: babel>=2.9 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (2.12.1)
Requirement already satisfied: alabaster<0.8,>=0.7 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (0.7.13)
Requirement already satisfied: imagesize>=1.3 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (1.4.1)
Requirement already satisfied: requests>=2.25.0 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (2.28.1)
Requirement already satisfied: packaging>=21.0 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (23.0)
Requirement already satisfied: importlib-metadata>=4.8 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from sphinx) (6.6.0)
Requirement already satisfied: zipp>=0.5 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from importlib-metadata>=4.8->sphinx) (3.15.0)
Requirement already satisfied: MarkupSafe>=2.0 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from Jinja2>=3.0->sphinx) (2.1.2)
Requirement already satisfied: charset-normalizer<3,>=2 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from requests>=2.25.0->sphinx) (2.1.1)
Requirement already satisfied: idna<4,>=2.5 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from requests>=2.25.0->sphinx) (3.4)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from requests>=2.25.0->sphinx) (1.26.13)
Requirement already satisfied: certifi>=2017.4.17 in /Users/felix.groll/Library/Python/3.9/lib/python/site-packages (from requests>=2.25.0->sphinx) (2022.12.7)
```

thp, 2023-05-09:
With these packages properly installed, you should have:

```
% ls -al /opt/homebrew/lib/libusb*
lrwxr-xr-x  1 thp  admin  52 19 Dez 16:56 /opt/homebrew/lib/libusb-0.1.4.dylib -> ../Cellar/libusb-compat/0.1.8/lib/libusb-0.1.4.dylib
lrwxr-xr-x  1 thp  admin  46 18 Aug  2022 /opt/homebrew/lib/libusb-1.0.0.dylib -> ../Cellar/libusb/1.0.26/lib/libusb-1.0.0.dylib
lrwxr-xr-x  1 thp  admin  40 18 Aug  2022 /opt/homebrew/lib/libusb-1.0.a -> ../Cellar/libusb/1.0.26/lib/libusb-1.0.a
lrwxr-xr-x  1 thp  admin  44 18 Aug  2022 /opt/homebrew/lib/libusb-1.0.dylib -> ../Cellar/libusb/1.0.26/lib/libusb-1.0.dylib
lrwxr-xr-x  1 thp  admin  42 19 Dez 16:56 /opt/homebrew/lib/libusb.a -> ../Cellar/libusb-compat/0.1.8/lib/libusb.a
lrwxr-xr-x  1 thp  admin  46 19 Dez 16:56 /opt/homebrew/lib/libusb.dylib -> ../Cellar/libusb-compat/0.1.8/lib/libusb.dylib
(m1) ~% 
```

However, I see from your log output that your Homebrew installation root seems to be `/Users/felix.groll/homebrew`. The script `scripts/macos/build-macos` assumes Homebrew for aarch64 is in `/opt/homebrew`.

If you open the script, at the beginning you should see this block:

```
case "$MACOS_CPU_ARCH" in
    x86_64)
        echo "Configuring for Intel architecture (x86_64)"
        MACOS_HOMEBREW_LIB=/usr/local/lib
        MACOS_HOMEBREW_INCLUDE=/usr/local/include
        ;;
    arm64)
        echo "Configuring for Apple architecture (arm64)"
        MACOS_HOMEBREW_LIB=/opt/homebrew/lib
        MACOS_HOMEBREW_INCLUDE=/opt/homebrew/include
        ;;
    *)
        echo "Unknown CPU architecture: $MACOS_CPU_ARCH"
        exit 1
        ;;
esac
```

If you change these lines:

```
        echo "Configuring for Apple architecture (arm64)"
        MACOS_HOMEBREW_LIB=/opt/homebrew/lib
        MACOS_HOMEBREW_INCLUDE=/opt/homebrew/include
```

You can make them point to `/Users/felix.groll/homebrew/lib` and `/Users/felix.groll/homebrew/include`, respectively. After that, use the build script to build it.

Alternatively, you can reinstall Homebrew into `/opt/homebrew` (the default installation path) and use the build scripts as-is.

thp, 2023-05-09:
I've updated the documentation to make this clear: 35399122c3f03406dc07b04a8d01858c8258e635

Please reopen if you still have troubles building the library.

commit 35399122 (2023-05-09): Docs: Homebrew installation path assumptions (Closes #487)
closed 2023-05-09 by thp