thp/psmoveapi #496: [HELP] Psmove controller not calibrating

AndrejPatak, 2024-09-26:
Hello, I've recently found my old psmove controller and wanted to see if I can use it for some fun on my arch linux laptop. I installed the arch package through yay, and connected the controller via usb.
First I did:
`sudo psmove pair
Connected controllers: 1
[PSMOVE WARNING] No calibration file found (/etc/psmoveapi/00_06_f5_18_f5_70.calibration or /etc/psmoveapi/00_06_f5_18_f5_70.calibration)
[PSMOVE WARNING] Magnetometer in 00:06:f5:18:f5:70 not yet calibrated.
PSMove #1 connected via USB.
[PAIRING LINUX] Found host adapter: 6c:6a:77:50:9a:41 (name=hci0)
[PAIRING LINUX] Running: 'systemctl stop bluetooth.service'
[PAIRING LINUX] Writing file: /var/lib/bluetooth/6C:6A:77:50:9A:41/00:06:F5:18:F5:70/info
[PAIRING LINUX] Writing file: /var/lib/bluetooth/6C:6A:77:50:9A:41/cache/00:06:F5:18:F5:70
[PAIRING LINUX] Running: 'systemctl start bluetooth.service'
Pairing of #1 succeeded!
Controller address: 00:06:f5:18:f5:70
Calibration data available and saved.`

Then I tried to calibrate it, but:
`sudo psmove calibrate
Connected controllers for calibration: 1
[PSMOVE WARNING] Magnetometer in 00:06:f5:18:f5:70 not yet calibrated.
Ignoring non-Bluetooth PS Move #0

Finished PS Move #0
`

Then I figured maybe I should run `psmove register 00:06:f5:18:f5:70`:
`sudo psmove register 00:06:f5:18:f5:70
[PAIRING LINUX] Found host adapter: 6c:6a:77:50:9a:41 (name=hci0)
[PAIRING LINUX] File /var/lib/bluetooth/6C:6A:77:50:9A:41/00:06:F5:18:F5:70/info is already up to date.
[PAIRING LINUX] File /var/lib/bluetooth/6C:6A:77:50:9A:41/cache/00:06:F5:18:F5:70 is already up to date.
Paired
`

I tried to calibrate again, but no luck:
`sudo psmove calibrate 00:06:f5:18:f5:70
Connected controllers for calibration: 1
[PSMOVE WARNING] Magnetometer in 00:06:f5:18:f5:70 not yet calibrated.
Ignoring non-Bluetooth PS Move #0

Finished PS Move #0
`

I dont know how to troubleshoot this nor do i know anything about the psmove controllers and how they work.
If I should provide more info, please ask, thank you for your time in advance

nitsch, 2024-09-28:
> Then I tried to calibrate it, but:
> `sudo psmove calibrate
> Connected controllers for calibration: 1
> [PSMOVE WARNING] Magnetometer in 00:06:f5:18:f5:70 not yet calibrated.
> Ignoring non-Bluetooth PS Move #0

It is telling you that controllers must be connected via Bluetooth for
calibration.

Since you successfully finished pairing previously, you should be able
to disconnect USB and establish a Bluetooth connection by pressing the
PS button on the controller.

thp, 2024-09-30:
I've improved the error message, as the "Finished PS Move #0" message might have been a bit misleading.

@AndrejPatak feel free to reopen if you can't get magnetometer calibration to work over Bluetooth.

closed 2024-09-30 by thp

AndrejPatak, 2024-09-30:
> It is telling you that controllers must be connected via Bluetooth for
> calibration.
>
> Since you successfully finished pairing previously, you should be able
> to disconnect USB and establish a [...]

I see now. I will try that when I get home, but I think my psmove's battery is completely dead, so I'm not sure if I'll be able to even use it. Do you know of any replacement batteries I could get somewhere?

commit 8cc40f73 (2024-09-30): magnetometer_calibration: Improve error message for USB connections (closes #496)