thp/psmoveapi #498: Unable to detect PS Eye Camera

steve-hops, 2025-01-09:
hi there, im trying to use the psmoveapi for a school project, however, my PS Eye Camera cant seem to be detected. im wondering is there something i might have missed out?

self.tracker = psmove.PSMoveTracker()
AttributeError: module 'psmove' has no attribute 'PSMoveTracker'

i checked that the ps eye camera works fine.
i using verision 4.0.12 of psmoveapi on ubuntu 22.04

thp, 2025-01-10:
Have you built the project with tracker support, and installed all dependencies (e.g. OpenCV)? Can you attach the full build log (do a clean rebuild if you haven't captured the log, so it really builds everything from scratch)?

sengyshon, 2025-01-12:
[Untitled.odt](https://github.com/user-attachments/files/18389951/Untitled.odt)
hi there, have attached the log.
my apologies for my ignorance, how should i enable the tracker?

Thank you!

steve-hops, 2025-01-13:
[Untitled.odt](https://github.com/user-attachments/files/18393604/Untitled.odt)

hi there, have attached the log.
my apologies for my ignorance, how should i enable the tracker?

Thank you!

steve-hops, 2025-02-04:
hi @thp , may i ask how do i build the project with tracker support?
also, i cant use the set_led function,im just wondering if i missed out any steps, i followed the instructions on psmoveapi.readthedocs.io/en/latest/ 

[image]

steve-hops, 2025-02-08:
hi @nitsch , thanks for responding, i used the 4.0.12 branch as i was facing the same problem as https://github.com/thp/psmoveapi/issues/495, i tried to use the master branch today and am still facing the same problem

build log:
[Untitled.docx](https://github.com/user-attachments/files/18719943/Untitled.docx)

nitsch, 2025-02-08:
Before going through the steps for building 4.0.12 with tracker support
... Can't you just use the Linux binaries that come with that release?

steve-hops, 2025-02-09:
hi @nitsch ,apologies for my ignorance, may i clarify "use the Linux binaries that come with that release"?
Thank you!

nitsch, 2025-02-09:
The psmoveapi releases
(https://github.com/thp/psmoveapi/releases/tag/4.0.12) come with
pre-built binaries for multiple platforms. Linux is also included.

The library files (.so) can be found in the libs/ subdirectory of the
respective archive, header files in the include/ subdirectory and so on.

It looks like libpsmoveapi_tracker.so depends on OpenCV 3.2 (you can use
the `ldd` command-line tool to check), so that would have to be
installed on your system in order to use tracking support.

thp, 2025-02-15:
Also, for the master branch, check out the scripts in https://github.com/thp/psmoveapi/tree/master/scripts -- these also build OpenCV and so should include tracker support. This even works for the master branch (and is what CI uses to verify the API can be built).